mik3y/usb-serial-for-android #97: Get Default Prober method not available

GaryDoo, 2015-01-20:
Hi, 
I've just linked the latest jar v0.1.0 into the project and I'm trying to use the getDefaultProber() method that should be contained with the UsbSerialProber class but it's not available for use.  It's present in the repository code but not in the jar.

I'm trying to connect with a japanese thermal printer so I'm pretty sure that this is important to trying to establish drivers etc.

If anyone has any tips I'd appreciate hearing thme

cpbridges, 2017-03-28:
Also have this issue during the .jar.

jubeira, 2017-03-28:
@GaryDoo @cpbridges the released version 0.1.0 is pretty old, and the API has changed since then.
You can either build the library from source and use the latest methods (I would do that), or go back to that release and check how the usage was before.

kai-morich, 2019-10-26:
use latest library

closed 2019-10-26 by kai-morich