mik3y/usb-serial-for-android #99: availableDrivers is empty

chackor, 2015-01-24:
This lines returns an empty list. 
            List<UsbSerialDriver> availableDrivers = UsbSerialProber.getDefaultProber().findAllDrivers(manager);

please advice how to address this.

Thanks,
R

victordiaz, 2015-02-18:
I have the same issue with some devices. It works for me with Arduino based ones, but not with mbed.
So far I know is not a Android issue since other USB monitoring apps can detect my device. 

Any idea?

Cheers

xiiaoleii, 2016-03-23:
me too. the result is: 0 device(s) found.

jonathanEp1, 2017-07-31:
Me too. did anyone solve this one?

hrshs, 2018-12-01:
https://github.com/mik3y/usb-serial-for-android#probing-for-unrecognized-devices

kai-morich, 2019-10-26:
use custom prober as mentioned in previous comment

closed 2019-10-26 by kai-morich