mik3y/usb-serial-for-android #102: No Device found: Arduino Nano V3.0

cracker0dks, 2015-03-11:
Hey, 
why does the example app dont show me my Arduino Nano V3.0 ? (0 devices found)
Im using a samsung galaxy s2 and the App "USB Host Controller" shows me an USB2.0-Serial device. 

I debuged it and the problem ist, that there are no drivers found for this device. (availableDrivers.isEmpty() is true)

Can anyone help ?
thanks :)

MRodriguess, 2015-04-08:
Hi I'm having the same problem with arduino uno. Did you managed to get any solution for that exception?

cracker0dks, 2015-04-08:
nop not yet... :(

MRodriguess, 2015-04-08:
Thanks..I will let you know if I found some solution for my case. Might help you.

cracker0dks, 2015-04-08:
that would be great. thanks !

MRodriguess, 2015-04-08:
Well I made some progress. At least for Arduino UNO I managed to get it pass trough that exception by adding it manually to the costumProbeTable. this.

    // Create a costumProbeTable based on the defaultProbeTable
    ProbeTable costumProbeTable = UsbSerialProber.getDefaultProbeTable();

And then add the device info
    costumProbeTable.addProduct(arduino.VID(), arduino.PID(), CdcAcmSerialDriver.class);

finally
    List<UsbSerialDriver> availableDrivers =           UsbSerialProber.getProber(costumProbeTable).findAllDrivers(mManager);

and loop that list to search for the driver, of if that's the only one you are using, get the first. Then initialize the connection.

It passed trough the initialization, but when I send some data from the arduino. It breaks and the USBRequestJNI goes crazy with close init close init .... and I can't get the data. I'm done for today. I will try tomorrow to investigate a bit more.

MRodriguess, 2015-04-09:
Hi cracker0DKS I have my arduino UNO working with android 4.03, using this library. I have updated the file CdcAcmSerialDriver to the last version and it worked. Are you using the last version?

cracker0dks, 2015-04-09:
29 Days ago i used the last version. glad that it worked for u now. I will try this if I can. Thanks :+1:

defvs, 2018-01-26:
Are you using an original arduino with an FTDI chip, or a clone with CH340G ? The latter isn't yet supported (see #194 )

Some more infos on differences between the two in #107

kai-morich, 2019-10-06:
CH340 is supported. The PRs mentioned in #194 are merged

closed 2019-10-06 by kai-morich