mik3y/usb-serial-for-android #105: native serial support (RS232)

mbrasile, 2015-04-08:
hi all. I come here trying to use <a href="https://github.com/xseignard/cordovarduino/issues/29#issuecomment-90822523">Cordovarduino</a> that rely on usb-serial-for-android.

I have a device with native serial port (/dev/ttyS0) and I was wondering if it is possible to add this kind of support to the project, to let Cordova drive directly a RS232 channel.

thank you in advance for your interest.

xseignard, 2015-04-09:
I would like to know if there is a way to use a native & internal serial port with your lib as stated @mbrasile ?

Thanks & regards,

Xavier

xseignard, 2015-04-21:
Hello there, any updates? /cc @mbrasile @mbrasile

xseignard, 2015-05-31:
Any updates???? /cc @mik3y

xseignard, 2015-07-06:
Hello any updates @mik3y @mbrasile ?

fabiomig, 2015-07-12:
Any updates? ETA?

yincrash, 2016-07-09:
AFAIK, there is no native serial port support in Android's SDK. There have been old projects that implemented serial port support by accessing the device via file `/dev/tty*`. See (here)[https://github.com/GinRyan/android-serialport-api]. This library is built around the USB host SDK in Android and probably would be awkward to shoehorn a native port in there.

mik3y, 2016-09-11:
This topic is covered in the [troubleshooting wiki](https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#how-do-i-access-devttyusb0); as yincrash says, that is not supported.

closed 2016-09-11 by mik3y