mik3y/usb-serial-for-android #107: ATMega328

gaara92, 2015-05-08:
Does this library has a chance to work with a ATMega328 chip (Arduino Nano) ?
Thanks

defvs, 2018-01-26:
Depends. Are you using a nano clone or an official one ? If you're using a clone, it surely has the CH340G chip, which is not compatible at the time (see issue #194 )
If you are using an official one, you'll be using the FTDI FT232 which is compatible
IF you are using the Atmega328 chip by itself, outside of a nano (ex : using the through-hole package that comes with a Uno outside of it) there won't be any chip for Serial to USB and you'll need one for communication. Be sure then to grab an FTDI one.

kai-morich, 2019-10-06:
CH340 is also supported

closed 2019-10-06 by kai-morich