mik3y/usb-serial-for-android #109: App can read from USB but has writing problems

andremacpereira, 2015-06-02:
I need to write several times repeatedly in order to send a single USB command of 8 bytes.

What could I possibly be missing?

Is there any writing example for CDC/ACM?

sureshjoshi, 2015-06-26:
Any luck with this?

sureshjoshi, 2015-06-26:
So, I was having a lot of trouble here, and flipping the endpoints and such, my device never seemed to be receiving data.

I tried the Physicaloid library (https://github.com/ksksue/PhysicaloidLibrary) and was able to get reading/writing working correctly. If I get some time (I know this won't be soon), I'm going to hook up a USB monitor and see what's up with the CDCACM endpoints.

kai-morich, 2019-10-26:
for various CDC/ACM devices like arduino you should set DTR+RTS

closed 2019-10-26 by kai-morich