mik3y/usb-serial-for-android #112: Incorrect Endpoint assignment in CDC Driver (with fix)

gnaur, 2015-07-07:
The assignment of endpoints to the mReadEndpoint and mWriteEndpoint variables in CdcAcmSerialDriver.open is incorrect.

The code assumes the read(in) endpoint is always endpoint(1).
This is not always true, the device can assign either endpoint as the in or out direction.

The code should dynamically assign the correct endpoint to the appropriate variable.

Symptom is the device appearing to always return a 16 byte buffer of nulls in response to a read.

A fix is below:
Replace lines CdcAcmSerialDriver.java 129 through 132:

---
               mReadEndpoint = mDataInterface.getEndpoint(1);
                Log.d(TAG, "Read endpoint direction: " + mReadEndpoint.getDirection());
                mWriteEndpoint = mDataInterface.getEndpoint(0);
                Log.d(TAG, "Write endpoint direction: " + mWriteEndpoint.getDirection());
---
+++
                UsbEndpoint ep0 =  mDataInterface.getEndpoint(0);
                UsbEndpoint ep1 =  mDataInterface.getEndpoint(1);
                if(ep0.getDirection() == UsbConstants.USB_DIR_IN) {
                    mReadEndpoint = ep0;
                    mWriteEndpoint = ep1;
                }
                else {
                    mReadEndpoint = ep1;
                    mWriteEndpoint = ep0;
                }

                Log.d(TAG, "Read endpoint direction: " + mReadEndpoint.getDirection());
                Log.d(TAG, "Write endpoint direction: " + mWriteEndpoint.getDirection());
+++

This patch should fix a number of bugs reported on this list.

Ralph

mik3y, 2015-07-07:
Sounds great. I don't have the cycles to prepare a change right now, but if someone would like to submit a pull request I can review it quickly.

thibautd, 2015-09-17:
I had this issue, didn't understand why write() didn't work. This fixed my issue, thanks.

kai-morich, 2019-10-06:
fixed with #166

closed 2019-10-06 by kai-morich