mik3y/usb-serial-for-android #114: Deactivate Debug Messages

Maikell84, 2015-07-09:
With every transaction made, the usb-serial-for-android is spamming my console, which makes debugging really hard.
Is there a way to deactivate these messages with a switch or something?

	02-17 09:30:21.590    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ close
	02-17 09:30:21.590    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ init
	02-17 09:30:21.594    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ close
	02-17 09:30:21.594    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ init
	02-17 09:30:21.598    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ close
	02-17 09:30:21.598    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ init
	02-17 09:30:21.602    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ close
	02-17 09:30:21.602    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ init
	02-17 09:30:21.606    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ close
	02-17 09:30:21.606    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ init
	02-17 09:30:21.610    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ close
	02-17 09:30:21.610    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ init
	02-17 09:30:21.614    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ close
	02-17 09:30:21.614    2332-2349/com.dummydomain.app D/UsbRequestJNI﹕ init

On Stackoverflow the same question remains unanswered, so I'm opening this issue.

mik3y, 2016-03-23:
Unfortunately `UsbRequestJNI` is coming from the underlying implementation in the Android system, not the library. :-(

closed 2016-03-23 by mik3y

ch00k17y, 2019-02-09:
`adb shell setprop log.tag.UsbRequestJNI SUPPRESS`  helps to deactivate UsbRequestJNI.

How to disable these messages?
02-09 23:13:05.148 22916-22975/... D/SerialInputOutputManager: Read data len=1
02-09 23:13:05.150 22916-22975/... D/SerialInputOutputManager: Read data len=2
...

A function to switch DEBUG off would be nice.

kai-morich, 2019-02-10:
you can deactivate with: `adb shell setprop log.tag.SerialInputOutputManager SUPRESS`

ch00k17y, 2019-02-10:
tried already but setprop throws the following error
```
130|shell@hammerhead:/ $ setprop log.tag.SerialInputOutputManager SUPPRESS
setprop: name 'log.tag.SerialInputOutputManager' too long; try 'log.tag.SerialInputOutputManage'
```

kai-morich, 2019-02-11:
adb setprop with long names does not work before Android 8.0. I typically filter out various unimportant tracing in Android Studio.