mik3y/usb-serial-for-android #118: FTDI driver improvements

uholeschak, 2015-09-19:
Hello,
have have added some improvements to FTDI serial driver:
- Allow to read and write the latency timer
- Set and get modem lines
- Fixed async read code (but is still disabled because it only works with newer android versions)

With the standard latency timer of 16 ms I sometimes lost packets during reception if the telegram was very long. In my code I have now changed this time to 50ms, with no packet loss any more.

For stable communication I had to increase the thread priority of my own SerialInputOutputManager to highest and changed the read timeout to 1s.

Could you add this patch to GitHub?

Ulrich
```
From 06e8c723ec3149e28637e5039fcdcbe18ec27edd Mon Sep 17 00:00:00 2001
From: uholeschak <ulrich@holeschak.de>
Date: Sat, 19 Sep 2015 09:45:55 +0200
Subject: [PATCH 1/2] Added missing functions: - latency timer - modem lines

---
 .../android/usbserial/driver/FtdiSerialDriver.java | 97 ++++++++++++++++++----
 1 file changed, 83 insertions(+), 14 deletions(-)

diff --git a/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java b/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java
index 1a9a66b..eb064b5 100644
--- a/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java
+++ b/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java
@@ -152,6 +152,11 @@ public class FtdiSerialDriver implements UsbSerialDriver {
          */
         private static final int SIO_SET_FLOW_CTRL_REQUEST = 2;
 
+        private static final int SIO_SET_DTR_HIGH = 0x0101;
+        private static final int SIO_SET_DTR_LOW = 0x0100;
+        private static final int SIO_SET_RTS_HIGH = 0x0202;
+        private static final int SIO_SET_RTS_LOW = 0x0200;
+
         /**
          * Set baud rate.
          */
@@ -166,6 +171,26 @@ public class FtdiSerialDriver implements UsbSerialDriver {
         private static final int SIO_RESET_PURGE_RX = 1;
         private static final int SIO_RESET_PURGE_TX = 2;
 
+        /**
+         * Get modem status.
+         */
+        private static final int SIO_GET_MODEM_STATUS_REQUEST = 5;
+
+        public static final int SIO_MODEM_STATUS_CTS = 0x10;
+        public static final int SIO_MODEM_STATUS_DSR = 0x20;
+        public static final int SIO_MODEM_STATUS_RI = 0x40;
+        public static final int SIO_MODEM_STATUS_RLSD = 0x80;
+
+        /**
+         * Set the latency timer.
+         */
+        private static final int SIO_SET_LATENCY_TIMER_REQUEST = 9;
+
+        /**
+         * Get the latency timer.
+         */
+        private static final int SIO_GET_LATENCY_TIMER_REQUEST = 10;
+
         public static final int FTDI_DEVICE_OUT_REQTYPE =
                 UsbConstants.USB_TYPE_VENDOR | USB_RECIP_DEVICE | USB_ENDPOINT_OUT;
 
@@ -183,7 +208,9 @@ public class FtdiSerialDriver implements UsbSerialDriver {
 
         private int mInterface = 0; /* INTERFACE_ANY */
 
[... 136 lines cut ...]
From 5312fdf49808c87b629faf20eaf21579b774bcb4 Mon Sep 17 00:00:00 2001
From: uholeschak <ulrich@holeschak.de>
Date: Sat, 19 Sep 2015 11:05:42 +0200
Subject: [PATCH 2/2] Fixed modem status

---
 .../main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java   | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java b/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java
index eb064b5..61155a2 100644
--- a/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java
+++ b/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java
@@ -287,7 +287,7 @@ public class FtdiSerialDriver implements UsbSerialDriver {
 
         public int getModemStatus() throws IOException {
             byte[] data = new byte[2];
-            int result = mConnection.controlTransfer(FTDI_DEVICE_IN_REQTYPE, SIO_GET_LATENCY_TIMER_REQUEST,
+            int result = mConnection.controlTransfer(FTDI_DEVICE_IN_REQTYPE, SIO_GET_MODEM_STATUS_REQUEST,
                     0, 0 /* index */, data, data.length, USB_WRITE_TIMEOUT_MILLIS);
             if (result != 2) {
                 throw new IOException("Get modem statusfailed: result=" + result);
-- 
1.9.5.msysgit.1
```

RePeet13, 2016-03-28:
@mik3y Improvements are generally good right?

uholeschak, 2016-03-28:
I already forgot about this patch,
many people are using the modified version now in my app, up to now with no complaints ...

kai-morich, 2019-11-23:
added your patches for latency timer and control lines:
21e96594d2b762ed1d9840c603d02410c422a57d
6b32e25e9c957726fd40a7158b44b9c009f0adfa

closed 2019-11-23 by kai-morich