mik3y/usb-serial-for-android #124: *FIX* Sending CMDS to Arduino

Rafcin, 2016-01-06:
So some of you may have issues were Arduino will not understand your packet data you sent, that is due to weird (A) symbol appearing before each command that was US-ASCII so to solve it, the port connection should be a separate function that is only called once, EX: onResume() and onPause() so keep it in mind, after its called and not looped your all set!

sogz2000, 2016-05-14:
Dear Rafcin, do you have experience to connect with smt32 board with interrupt mode, When I post comand to USB device with UsbRequest.quence , it return true in most of the time, but the board seems not received the command and no feedback, its devide info as below:,
<usb-device vendor-id="1155" product-id="22352"/>

thanks a lot

mik3y, 2016-09-11:
Please try [the google group](https://github.com/mik3y/usb-serial-for-android#help--discussion) for help.

closed 2016-09-11 by mik3y