mik3y/usb-serial-for-android #125: No read message with Nexus 5x + Ftdi driver.

bidasknakayama, 2016-01-08:
Hi All.
I need your advice or comment to resolve a issue.

I have a Serial Device (Ftdi).  [OTG cable]-----[USB]-[RS485]-----------[A Device] 
With my Nexus 7 , My software can received data from the Device in a second.

and, I tried with Nexus 5x with same Android source code.
I can receive something. but This is just Haphazard data only.
CRC check does not pass at all.
I run 10-20 minutes , One time, I received data after passing CRC check !
but It will not happen again... never.

I know To resolve this issue, Information is not enough at all.
1) Read time out maybe shorter. I tried to increase timeoutMillis on FtdiSerialDriver.java (Write, Read).
 but this does not change anything.

2) Nexus 5C is USB-C OTG cable. Nexus 7 is USB-OTG cable. 
  Amazon will send me another cable to me. I bought again...for test.

3) Nexus 7 is running USB-MTP connection.
   I tried USB-MTP, PTP on Nexus 5x too. but this is not answer.

4) Baudrate. Device maker suggest to us 115200. I change this too. 
but this is not answer.

At least, I receive 1 data for my 1 days test, 
This mean, If I change something a bit. There is possibility to receive data.

Please give us a advice.. I will try...

Kazuhiko Nakayama.

bidasknakayama, 2016-01-13:
I find 

    java.io.IOException: Expected at least 2 bytes

at here .

    private final SerialInputOutputManager.Listener mListener =
    new SerialInputOutputManager.Listener() {
    @Override
    public void onRunError(Exception e) { 

What can I do ?

bidasknakayama, 2016-01-13:
Here is the data that I received.

I added some indent on header of data to format.
THis mean, When I receive data, Header or Tail is lost...
Do you have any idea ?

in my system, Header is 0xAA. 
each 1 line is 1 received packages.

```
               44 47 32 20 50 4d 31 34 30 33 20 76 31 2e 39 20 41 70 72
                  20 32 33 20 32 30 31 33 20 30 78 43 37 44 45 00 a7
aa 28 49 30 42 44 47 32 20 50 4d
                     31 34 30 33 20 76 31 2e 39 20 41 70 72 20 32 33 20 32 30 31 33 20 30 78 43 37 44 45 00 a7
aa 28
      49 30 42 44 47 32 20 50 4d 31 34 30 33 20 76 31 2e 39 20 41 70 72 20 32 33
                                                                     20 32 30 31 33 20 30 78 43 37 44 45 00 a7
aa 28 49 30 42 44 47 32 20
               50 4d 31 34 30 33 20 76 31 2e 39 20 41 70 72 20 32 33 20 32 30 31 33 20 30 78 43 37 44 45 00 a7
aa 28 49 30 42 44
      47 32 20 50 4d 31 34 30 33 20 76 31 2e 39 20 41 70
                                                         72 20 32 33 20 32 30 31 33 20 30 78 43 37 44 45 00 a7
aa 28 49 30
aa 28 49 30 42
   44 47 32 20 50 4d 31 34 30 33 20 76 31 2e 39 20 41 70 72 20 32 33 20 32 30 31 33 20 30
                                                                                          78 43 37 44 45 00 a7
```

bidasknakayama, 2016-01-20:
Is there any way to set a Handshake parameter ?
    port.setParameters(int baudRate, int dataBits, int stopBits, int parity)
Is this only ?

bidasknakayama, 2016-01-21:
I tried to connect received data to large array. but it does not work as good data.
I gave up this issue. I do not recommend to use Nexus 5x yet with USB-Serial-for-Android.
Nexus 7 is working very good : )

aruntej, 2016-08-30:
Cool.

aruntej, 2016-08-30:
Is there somewhere i can find your code.

closed 2019-11-01 by kai-morich