mik3y/usb-serial-for-android #128: SerialInputOutputManager.Listener BufferSize for onNewData

barisozanguler, 2016-01-19:
Hi,

I'm communicating via CP210x usb device.

I want to change read each time 1 byte (BufferSize) for listener.   

```
private final SerialInputOutputManager.Listener mListener =
            new SerialInputOutputManager.Listener() {

                @Override
                public void onRunError(Exception e) {
                    Log.d(TAG, "Runner stopped.");
                }

                @Override
                public void onNewData(final byte[] data) {
                   
                }

            };
```

How can i do that ? 

Thanks.

fghjhuang, 2016-08-19:
i am looking for a way too,did u solve this problem?

barisozanguler, 2016-08-22:
No man. I'm process each byte with for loop. :( I was tried configurate somethings in library code but its not work.

fghjhuang, 2016-08-30:
ths anyway, i am doing it in a loop ,too

kai-morich, 2019-10-06:
the SerialInputOutputManager reads all available data, then you can split the data in a loop, as you also suggested

closed 2019-10-06 by kai-morich