mik3y/usb-serial-for-android #129: How to import this library in android studio?

XSparter, 2016-03-07:


aruntej, 2016-08-30:
Did you have any luck with importing this library to android studio

XSparter, 2016-08-30:
No... But i have maked a serial conenction in android.

Mushtaq123, 2016-11-07:
By Clicking Import Module by **File --> New-->Import Module** then Browse to your Library Project and Import

atrvicky, 2017-02-26:
After importing as @Mushtaq123 said, include this in Gradle and sync files
compile project(':usbSerialForAndroid')

Harin-Kaklotar, 2017-11-02:
download this project

open your android studio project

File > New > Import Module > set path of **usbSerialForAndroid** > Finish

rsc9421, 2018-02-20:
I have this error
Error:Application and test application id cannot be the same: both are 'com.hoho.android.usbserial.examples' for debugAndroidTest

buyukyukselburak, 2018-03-07:
I have this error
Error:Application and test application id cannot be the same: both are 'com.hoho.android.usbserial.examples' for debugAndroidTest
++

kai-morich, 2018-03-07:
please try with commenting out 'testApplicationId ...' in usbSerialExamples/build.gradle

alkocher, 2018-07-12:
Android Studio does not see the module

jjavierk, 2018-09-15:
1. Create your project, 
2. Then, File -> New -> Import Module and add the path of the folder usbSerialForAndroid.
3. Open settings.gradle and add to the include ':usbSerialForAndroid'
4. Open Modular settings (right click on app folder), 
5. Go to the Dependencies tab, click on the green '+' symbol -> Module dependency.
6. If you have done everything correctly, usbSerialForAndroid will be listed already, select it and press ok.
Check the image for the steps.
[image]

uklertis, 2019-03-28:
I have the same issue with @XSparter. Has anybody solved it?

kai-morich, 2019-10-26:
instead of importing the source in your project you can use gradle dependency as shown in README

closed 2019-10-26 by kai-morich