mik3y/usb-serial-for-android #131: New release

mik3y, 2016-03-23:
Make a new release & upload to maven.

RePeet13, 2016-03-23:
Do you have an eta on this? I have no notion how easy/long the process is to upload a new version to maven is. After that there should be a version bump in the gradle of kegtab, and an update published to the play store, right?

RePeet13, 2016-03-28:
@mik3y My poor kegerator is waiting to get its very own kegbot behind this issue, and updating the kegtab with the new library. Can you spare some time soon?

RePeet13, 2016-04-11:
@mik3y any movement on this issue?

RePeet13, 2016-04-19:
@mik3y Getting anywhere? It's been about a month.

mik3y, 2016-05-24:
@RePeet13 sorry, I've had my hands full. But I have once again been re-united with my USB gear. I'll try to close this out this week.

RePeet13, 2016-05-24:
👍

RePeet13, 2016-07-06:
@mik3y ready to shipit?

RePeet13, 2016-08-02:
@mik3y had a chance to package and upload yet?

James-Firth, 2018-02-12:
Also wondering if the maven repo will be updated soon :)

Is the process cumbersome for you @mik3y or just haven't had a chance? If it's cumbersome I'm sure someone would be willing to help make it easier for you

kai-morich, 2019-10-26:
now available on jitpack. see badge in README

closed 2019-10-26 by kai-morich