mik3y/usb-serial-for-android #143: Impossible to communicate with Nucleo F401RE

echo22, 2016-06-23:
Hi, I'm trying to communicate with an STM Nucleo F401RE via usb-serial-for-android but without results.
Everything works grat when I use an arduino, but have problems with Nucleo.
I think the problems is inside CdcAcmSerialDriver.java. In the method open the code look for 2 interfaces, the first is a control interface, the second is a data interface.
With Arduino is ok, but Nucleo has 4 different interfaces. This is the list of interfaces with endpoints (128 is Input, 0 is Output).
Interface 0`
D/Endpoint no 0: 128
D/Endpoint no 1: 0
D/Endpoint no 2: 128

Interface 1
D/Endpoint no 0: 128
D/Endpoint no 1: 0

Interface 2
D/Endpoint no 0: 128

Interface 3
D/Endpoint no 0: 0
D/Endpoint no 1: 128`

I tried to change the code in order to use different interfaces and endpoints but without luck. Someone has suggestions? Do you think this is the problem?
In another thread mobilinkd say that his board can work only at 9600. I need to work at 115200. https://github.com/mik3y/usb-serial-for-android/issues/126

Thank you.

atomsndbytes, 2016-06-23:
Just thinking , i had the same problem while using a F4 discovery board , then i realised that it was not working because the peak voltage for the UART line on F4 was 3.3V ,But for some reason the arduino Mega's +5V UART was working , so i added a +5V pull up on the 
UART line and made the UART passive , by changing the pin definitions to make the TX switch between low and floating.Will try adding some connection diagrams

echo22, 2016-06-24:
You think the problem is the voltage? It seems strange, when it's connected on the pc's usb it works well.

kai-morich, 2019-10-06:
with thic change 608c67499a10ccf5c25811e9d1611187192aaa45 the correct endpoints are used

closed 2019-10-06 by kai-morich