mik3y/usb-serial-for-android #153: issue reading FT234XD

yyf, 2016-09-15:
Been having issue reading FT234XD. 

1. Tried building the latest source using gradlew 
2. followed the steps in the release page
3. put usbSerialForAndroid-release.aar in my lib folder
4. added VID and PID to device filter xml
5. tried compile in Android Studio and had the following error:

Error:(194, 32) error: no suitable constructor found for SerialInputOutputManager(UsbSerialDriver,Listener)
constructor SerialInputOutputManager.SerialInputOutputManager(UsbSerialPort,Listener) is not applicable
(actual argument UsbSerialDriver cannot be converted to UsbSerialPort by method invocation conversion)
constructor SerialInputOutputManager.SerialInputOutputManager(UsbSerialPort) is not applicable
(actual and formal argument lists differ in length)
Error:(213, 26) error: cannot find symbol method setWriteBufferSize(int)
Error:(218, 30) error: cannot find symbol method write(byte[],int)
Error:(236, 26) error: cannot find symbol method setWriteBufferSize(int)
Error:(241, 30) error: cannot find symbol method write(byte[],int)
Error:(259, 26) error: cannot find symbol method setWriteBufferSize(int)
Error:(264, 30) error: cannot find symbol method write(byte[],int)
Error:(281, 26) error: cannot find symbol method setWriteBufferSize(int)
Error:(286, 30) error: cannot find symbol method write(byte[],int)
Error:(303, 26) error: cannot find symbol method setWriteBufferSize(int)
Error:(308, 30) error: cannot find symbol method write(byte[],int)
Error:(325, 26) error: cannot find symbol method setWriteBufferSize(int)
Error:(330, 30) error: cannot find symbol method write(byte[],int)
Error:(347, 26) error: cannot find symbol method setWriteBufferSize(int)
Error:(352, 30) error: cannot find symbol method write(byte[],int)
Error:(421, 40) error: cannot find symbol method acquire(UsbManager)
Error:(433, 30) error: cannot find symbol method open()
Error:(437, 34) error: cannot find symbol method close()
Error:(447, 30) error: cannot find symbol method setBaudRate(int)
Error:(453, 26) error: cannot find symbol method setWriteBufferSize(int)
Error:(458, 30) error: cannot find symbol method write(byte[],int)
Error:(498, 30) error: cannot find symbol method close()
Error:Execution failed for task ':app:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

Any suggestion or doc I can follow along? any plan for new release of usb-serial-for-android? Thanks.

yyf, 2016-09-16:
after adding VID/PID to device filter xml, able to get a prompt whenever FT234XD is connected, but UsbSerialDriver.getDevice() doesn't return anything. What are the steps to support a new device?

closed 2016-09-19 by yyf

nmilcoff, 2019-01-15:
@yyf would you mind sharing how did you solve this? I need to support FT234XD now and it doesn't seem to be possible