mik3y/usb-serial-for-android #155: PL2303 send 4 bytes,but APP read one by one

junerver, 2016-11-11:
[image]

when i use ComAssistant in PC read PL2303 Device , ComAssistant show **FF 01 60 61**.

fghjhuang, 2017-03-21:
yes me too, the result i read in a loop

fghjhuang, 2017-03-21:
just find in wiki：
When reading a message, why don't all bytes arrive at the same time?

The protocol layer does not guarantee all bytes will arrive in a single message. In fact the protocol layer doesn't have any knowledge of what your "message" is — it is just an interface for sending a serial stream of bytes.

For example, to receive a 100 byte string, you might read 64 bytes, then 36 bytes, instead of a single message of 100 bytes.

You need to account for this problem when designing your protocol. Some common techniques:

    Fixed length messages: If you a message is always going to be 100 bytes, just keep reading until you have all 100.
    Length-prefixed messages: Prefix every message with a fixed-length size value; your message is complete after you've read size more bytes.
    Newline-terminated messages: Read until you see a \n (or any other "terminal" character).

junerver, 2017-03-24:
thinks for you help

closed 2017-03-24 by junerver