mik3y/usb-serial-for-android #157: FindAllDriver returns empty driver list

goelankita, 2016-12-01:
Hello,

I am trying to connect to a hardware device which my organization build. I provided its VID and PID in xml files, but application is not able to detect my device. When I connect my device via USB, I got a messagebox asking that open this USB device by "SeialApp" and also USB device connected to phone. It means phone is detecting the device.

I tried to debug. "findAllDrivers" method is not returning any driver. I tried to connect a flash drive, my phone was able to open its file, but Serial USB application could not detect its driver too.

Please help. Any help will be appreciated.

Thanks so much in advance

akshaytaru2007, 2017-06-07:
I am facing the same issue

buyukyukselburak, 2018-03-07:
Could you solve this problem, if you success it, can you explain?

akshaytaru2007, 2018-03-20:
Please do tell.

14331195, 2018-04-28:
I don't know how to solve this problem. Please help.

hrshs, 2018-12-01:
I solve this problem by adding this
https://github.com/mik3y/usb-serial-for-android#probing-for-unrecognized-devices

josealonso, 2019-01-20:
> I solve this problem by adding this
> https://github.com/mik3y/usb-serial-for-android#probing-for-unrecognized-devices

Me too, thanks for the suggestion.

kai-morich, 2019-10-26:
use a custom prober as recomended in the 2 previous comments

closed 2019-10-26 by kai-morich