mik3y/usb-serial-for-android #159: UsbDeviceConnection.requestWait() never times out, so all reads are blocking on API 17+

micolous, 2016-12-29:
[Due to Android bug 39522](https://code.google.com/p/android/issues/detail?id=39522), it is not possible to use `UsbDeviceConnection.requestWait()` with a timeout.

At least the `CdcAcmSerialDriver` uses this on API Level 17+ to do a `read`.  This effectively makes all read operations blocking, even if the timeout is reached.

It would be useful to be able to explicitly specify `mEnableAsyncReads = false` to be able to always fall back to using `bulkTransfer`, which actually allows timeouts to be used.

carl1990, 2017-12-15:
where the var mEnableAsyncReads I can found and set

kai-morich, 2019-10-06:
0d48ed04e7cf3ff83c8089d4dedf61f1ba7277be always uses async read, as bulkTransfer can cause data loss. Blocking requestWait() are explicitely canceled on connection close. Blocking is not an issue, as reading should anyway be done in a separate thread.

kai-morich, 2019-10-26:
nevertheless the functions should not block unexpectedly.
requestWait.requestWait() supports timeout since Android 8.0
on older Android versions should throw UnsupportedOperationException

kai-morich, 2019-11-15:
read timeout is supported by all drivers with e2e9df846386ccdbf6625c057fd1cd5f94c16d41

closed 2019-11-15 by kai-morich