mik3y/usb-serial-for-android #172: how can i connect serial port with service from Broadcastreceiver

ctpxyza, 2017-05-31:
i need to close serial connection and reopen the connection from background, by after i receive usb state from broadcastreceiver, then use service to open connection, "can't connect" always shows to me. but if i start service from activity which register meta-data, it works will.
so , can you help me to let it works in the background. thx.

HopeForWholeVillage, 2018-11-22:
i meet the same problem，it‘s must work whit activity，the connection will be null when i opendevice in Broadcastreceiver,i aslo regist the meta-data and the action

ctpxyza, 2018-11-22:
> i meet the same problem，it‘s must work whit activity，the connection will be null when i opendevice in Broadcastreceiver,i aslo regist the meta-data and the action

i have found the reason of this issue, android usbhost in framework intercepts connection event and only pass the event to Activities which register the meta-data, then show a dialog to choose one of them.

closed 2019-10-06 by kai-morich