mik3y/usb-serial-for-android #173: CdcAcmSerialDriver port.read freezes thread

aleksandrmozgovoi, 2017-05-31:
If Ch34xSerialDriver.class then
Log.d("test",name + "[USB] step before READ");//run ok
readCount = port.read(bufRxTemp,100);
Log.d("test",name + "[USB] step after READ");//run ok
-> no problem

If CdcAcmSerialDriver.class then
Log.d("test",name + "[USB] step before READ");//run ok
readCount = port.read(bufRxTemp,100);
Log.d("test",name + "[USB] step after READ");//**Never run PROBLEM**

The device USB does not send anything.

Andre-Rb, 2017-06-05:
I got the same issue, it's making me mad :(
I investigating into lower Serial read frequency, or shorter strings.
It works with the simple arduino example tho'

Andre-Rb, 2017-06-05:
I got news, for me, it may be related to the use of `abort()` in my setup code

aleksandrmozgovoi, 2017-06-05:
I solved the problem. The read function in cdcAcmSerialDriver.class is similar to Ch34xSerialDriver.class. The whole fault is 'if (mEnableAsyncReads)'.

Andre-Rb, 2017-06-05:
Well for me, it was the abort() calling.

closed 2019-10-06 by kai-morich