mik3y/usb-serial-for-android #179: Read USB Joystic input

gunnicus, 2017-08-15:
Hi I connected my usb logitech joystick to my android phone, I'm trying to get the joystick input using usb-serial-for-android. At the moment the app shows '0 devices found'. What should I do? Is there a driver for the joystick?

olimaticer, 2017-08-31:
A joystick is a Human Input Device HID.
Use the activity event onKeyDown().

Maybe test it with a test version of RFO-Basic first.
The command is InKey$ key$, **keyEvent$**
[https://www.tapatalk.com/groups/rfobasic/hot-summer-of-code-t4408-s60.html#p35494](url)

olimaticer, 2018-04-25:
Maybe I did not explain it good enough.
A joystick is a Human Input Device HID.
You do not need usb-serial-for Android!
Use the activity event, maybe  onKeyDown(), onKeUp() etc..

kai-morich, 2019-10-06:
this library only supports serial-to-usb converts, not HID devices

closed 2019-10-06 by kai-morich