mik3y/usb-serial-for-android #186: Repo maintenance

jubeira, 2017-09-27:
PRs and issue count seem to keep growing over time, and it seems that there are plenty of users of this code around.

Has anyone been able to contact the owner/ maintainer? It would be awesome if this repo could be hosted by someone else with some time to maintain it and merge relevant contributions.

pavi2410, 2017-10-13:
I agree with you. This library is indeed very useful and should be maintained to support more features.

pavi2410, 2017-10-13:
How about if we fork this repo and maintain it, if @mik3y is not active anymore?

mik3y, 2017-10-13:
Hey folks! I would definitely be happy to bring in some new maintainers, as pretty clearly I don't have the time (or the hardware gizmos) necessary to maintain things. Would be happy to see it live on with some better stewardship..

As there's some incomplete work in `master` & bunch of dangling issues, I'll try to spend some time on maintenance this weekend and see where we land.

(Of course, you should always feel free to fork!)

Fyb3roptik, 2017-10-19:
Any update? I would love to have merged PR's please! I need to use this for a Mega. Will it work?

pavi2410, 2017-11-28:
Can anyone help me in merging PRs if I create a fork? I will you as collaborator.

mik3y, 2017-11-29:
Please see #191 !

closed 2017-11-29 by mik3y