mik3y/usb-serial-for-android #191: New maintainer wanted

mik3y, 2017-11-29:
Hi folks! I'm looking for someone to take over as maintainer for this project. As you can see, I haven't had the time to do so. (And I no longer have the devices to test, either!)

The new maintainer would have responsibility over reviewing submitted PRs, triaging issues, improving the code and publishing new releases. Ideally, you've already submitted PRs and/or maintain a fork, or have a track record from another open source project.

If you're interested, please leave a note below.

kai-morich, 2018-02-18:
I can imagine taking over this project

mik3y, 2019-08-09:
Hey @kai-morich - are you still interested in taking over? Let's give it a shot if so!

kai-morich, 2019-08-09:
yes, please send mail for further steps

eclectice, 2019-08-09:
I vote for kai ;)

kai-morich, 2019-08-15:
you can reach at mail@kai-morich.de

mik3y, 2019-10-03:
I've add you as a collaborator! If you'd like me to look at any pull requests, please poke me on them. Otherwise, feel free to merge as you see fit.

closed 2019-10-03 by mik3y

kai-morich, 2019-10-04:
I will merge the stuff I collected over time and will have a look at the rest.

I assume this library has LGPL license because this licences is used by libFTDI.
But what is the reason for Google Copyright 2011-2012?