mik3y/usb-serial-for-android #194: Request : Add support for CH340G

defvs, 2018-01-13:
The CH340G Usb-to-TTL device is included today in many Arduino clones out there. In fact it is cheap, works fine in most situation, but disadvantage is that software support is required (drivers for Windows for example)

By adding it, you will definitely support all Arduino out there, including unofficial ones.

If you want some clues, I think the app ArduinoDroid supports it.

AliensInc, 2018-01-26:
Agree, will be very useful

defvs, 2018-01-26:
Looks like there was some support but it doesn't seem to work well : see PRs #121 , and some more PRs are ready to merge like #195 #188 #156

defvs, 2018-01-26:
Not working for me atleast

kai-morich, 2018-02-09:
CH340G works for me with #121 included in master branch, the other PRs are extensions to support more options

kai-morich, 2019-10-06:
PR #121 #195 #188 #156 are merged

closed 2019-10-06 by kai-morich