mik3y/usb-serial-for-android #197: sport.setparameters problem

feherbt, 2018-03-21:
Hi,

I have got an Acer B3 A40FHD tablet and an EMPv800 coin acceptor via CP2102.

When I start the application everything works fine.
If I restart the app I have got an error at this line:
sPort.setParameters(9600, 8, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE);

The error in logical:
W/SerialInputOutputManager: Run ending due to exception: Attempt to invoke virtual method 'int android.hardware.usb.UsbEndpoint.getAddress()' on a null object reference
       java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.usb.UsbEndpoint.getAddress()' on a null object reference
           at android.hardware.usb.UsbDeviceConnection.bulkTransfer(UsbDeviceConnection.java:214)
           at android.hardware.usb.UsbDeviceConnection.bulkTransfer(UsbDeviceConnection.java:196)
           at com.hoho.android.usbserial.driver.Cp21xxSerialDriver$Cp21xxSerialPort.read(Cp21xxSerialDriver.java:185)
           at com.hoho.android.usbserial.util.SerialInputOutputManager.step(SerialInputOutputManager.java:159)
           at com.hoho.android.usbserial.util.SerialInputOutputManager.run(SerialInputOutputManager.java:141)
           at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:428)
           at java.util.concurrent.FutureTask.run(FutureTask.java:237)
           at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
           at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
           at java.lang.Thread.run(Thread.java:761)

Regards,
Ferenc

kai-morich, 2018-03-21:
might be an android lifecylce usage issue. maybe your connection was closed, but the SerialInputOutputManager was not stopped in onPause/onStop

feherbt, 2018-03-22:
you were right!!!! :)

closed 2019-10-06 by kai-morich