mik3y/usb-serial-for-android #200: Returns null "manager.openDevice" with Prolific PL2303

akdrmrk, 2018-04-16:
I can use ftdi driver successfully but the "openDevice" returns null everytime with Prolific PL2303.

What can be cause of this?

kai-morich, 2018-04-16:
you get null if usbManager.requestPermission(..) is required. maybe you have done this for FTDI device some time ago, but now have to repeat that for the PL2303

akdrmrk, 2018-04-17:
I used product id which return from driver.getDevice() but I noticed that, this is wrong product id. The proble is solved when adding all product id of prolific in device-filter

kai-morich, 2018-05-11:
please close issue if resolved

closed 2019-10-26 by kai-morich

mehrhossin, 2021-08-28:
"The proble is solved when adding all product id of prolific in device-filter"
Can you tell me exactly what you added?