mik3y/usb-serial-for-android #201: Unable to read from usb port, writing works well (FTDI).

greblus, 2018-05-03:
Hi guys.

I've been using usb-serial-for-android for a few years now in my port of AspeQt - Atari Serial Peripheral Emulator for Android and it worked very well, but I started getting reports from users saying that despite their device fully supports USB Host AspeQt doesn't work. I couldn't reproduce this problem, but finally I got two such devices, both with Android 7.1 where I observe similar behavior.

What I've been able to conclude till now:

1. Probably it's not due to missing permissions, in debug mode it shows they're granted:
USB     : Device found!
USB     : Received permission result OK
USB     : Device OK
USB     : Driver found for attached usb device
D FtdiSerialDriver: claimInterface 0 SUCCESS
USB     : Device opened
USB     : setBaudrate: 19200

2. AspeQt emulates cassette images loading and it's working, possibly because it's only writing data to the usb port, without waiting for any acks or whatever, so the communication is one way.

3.I also prepared an experimental version which d2xx library and is working - but it's much slower, buggy and closed-source, so I don't want to use it. But it's a good test for a user to make sure if the device is fully supporting usb-host (otg).

Now the hard part - what do you suggest to check? I've been fiddling around with endpoints, indexes, no luck so far. The symphtoms are - no data incoming from ftdi. Same code works well on my LG X Power, with 6.0 and Lolipop based tablets. Have you experienced something similar?

Cheers,
W.

matthel, 2018-08-07:
I am facing the same issue on different smartphone running Android 7.1. The CdCAcmSerialDriver.read function never return any bytes.
Have you any update on that?

Thank you.

greblus, 2018-08-07:
Unfortunately after spending really a lot of time on it I haven't solved this issue yet. I'll have to create a standalone debug app without Qt dependency to make testing less time consuming.

dheeraj12190, 2018-11-26:
I am facing the same issue do you have any solution for this.

greblus, 2018-11-26:
No. Unfortunately all my attempts failed so far and I've no idea why it's not working on newer Android versions while ftdi's d2xx library is somewhat working.

kai-morich, 2018-11-26:
please try with kai-morich/usb-serial-for-android fork, maybe commit fcd8596bdd3c267731fc40b19818757399c73241 (enable async read for FTDI ...) solves your issue

kai-morich, 2018-11-27:
please try with 
        port.setDTR(true);
        port.setRTS(true);
after port.setParameter().
if the other side has some kind of flow control implemented, it might not send data w/o these control lines, e.g. arduino needs this

another option is using 
        ioManager = new SerialInputOutputManager(port, this);
        Executors.newSingleThreadExecutor().submit(ioManager);
I never used read in UI thread

dheeraj12190, 2018-11-27:
@kai-morich Thanks for your support and prompt response.
this one is not working 
port.setDTR(true);
port.setRTS(true);
I am new to android can you elaborate second option. Thanks in advance.

greblus, 2018-11-27:
Yeah. I've also just backported your [fcd8596](https://github.com/mik3y/usb-serial-for-android/commit/fcd8596bdd3c267731fc40b19818757399c73241) to my fork and it doesn't help. Writing to port is working without issues so [loading a cas image](https://drive.google.com/open?id=1DbRfIIFscP8lzTaMhZK9VN5W4IK_nea7) using my Android tv box as a "casette player" for 8-bit Atari is ok ;) but reading from port is not working, so disk emulation is non-functional.

kai-morich, 2018-11-27:
SerialInputOutputManager is a class in usb-serial-for-android library. Please look at the example for usage

dheeraj12190, 2018-12-01:
@kai-morich I have gone through the example. I have complied that it is running on my mobile. but when I used that in my app it is not going in OnNewData Event belowis my code 

    private final ExecutorService mExecutor = Executors.newSingleThreadExecutor();
    private SerialInputOutputManager mSerialIoManager;
    private final SerialInputOutputManager.Listener mListener = new SerialInputOutputManager.Listener() {
        @Override
        public void onNewData(final byte[] data) {
            MainActivity.this.runOnUiThread(new Runnable() {
                @Override
                public void run() {
                    MainActivity.this.updateReceivedData(data);
                }
            });
        }

        @Override
        public void onRunError(Exception e) {

        }
    };
    private void updateReceivedData(byte[] data) {
        final String message = "Read " + data.length + " bytes: \n"  + HexDump.dumpHexString(data) + "\n\n";
        Toast toast = Toast.makeText(getApplicationContext(),message, Toast.LENGTH_SHORT);
        toast.setDuration(10);
        toast.setMargin(50,50);
        toast.show();
    }

 
            port.open(connection);
            port.setParameters(9600, 8, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE);

            byte buffer[] = new byte[16];
            byte disc[] = {0x7E, (byte)0xA0, 0x07, 0x03, 0x61, 0x53, 0x65, (byte)0x81, 0x7E};
            port.write(disc,100);
            mSerialIoManager = new SerialInputOutputManager(port, mListener);
            mExecutor.submit(mSerialIoManager);

I don't know either it is Right or not. can you please help to make the things working thanks in advance.