mik3y/usb-serial-for-android #204: I can read 61 byte via "Prolific PL2303 cable.

akserhat, 2018-05-21:
I am writing of incoming response after the process of writing data to serial port . Read-out always can be 61 byte.
 When ı reading again ı can read next 61 byte.(sometimes all of them turn 0)

The problem is that, ı have to read 2000 byte but in the middle of the 2000 byte again read the 61 byte 

1-why can ı read 61 byte
2-why I can not start reading begining

kai-morich, 2018-06-09:
the chunk size depends on the used chip and baud rate.

eclectice, 2019-01-17:
It is better to design a transport protocol to carry the payload with a start marker and an end marker. Use those markers in the receiving loop to capture the payload.

kai-morich, 2019-11-01:
the number of bytes received in one request is not predictable, see https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#when-reading-a-message-why-dont-all-bytes-arrive-at-the-same-time

closed 2019-11-01 by kai-morich