mik3y/usb-serial-for-android #208: USB-C

rsm123, 2018-08-15:
I tried connecting FT232 module to Android phone USB-C port and the program is unable to detect the FTDI driver. In phones with micro USB port it autodetects when using with OTG cable. Is this is a known issue with USB-C phones?

eclectice, 2019-01-17:
USB-C physical standard is USB agnostic: It should be able to support USB-2.0 on FT232R chip. 

But, USB-C connector can only mate which each other and it cannot mate with everything else like Micro-USB connector. Probably, you need an adapter between Micro-USB and USB-C connector to match the signal. 

But in your case, you use an OTG cable which is probably not matching the signal correctly. You should try to use a non-OTG adapter like this https://www.belkin.com/us/p/P-F2CU058/

Or maybe, the supplier of your FT232R that uses USB-C doesn't fully comply the necessary signals. As a reference: https://www.embedded.com/electronics-blogs/benson-s-blocks/4442214/USB-Type-C-in-a-Micro-B-world

kai-morich, 2019-10-06:
works with USB-C to USB-A converter that comes with Samsung phones. You should try with different connector/cable

closed 2019-10-06 by kai-morich