mik3y/usb-serial-for-android #210: How to connect to a serial port with this library

BrotherV, 2018-10-14:
Hi, I appreciate your work. Unfortunately I couldn't find a right way to connect to serial port, There are two different ways that you mentioned. first of them relate to jar library and second one relate to the last commit. In the first sample Jar file doesn't support any of control values such as stop bit and so on. Can you guide me to fix it? Even I can't connect to an Arduino.

Sample code:
UsbSerialProber prober = UsbSerialProber.getDefaultProber();
UsbSerialDriver driver = prober.probeDevice(device);
if(driver != null){
	List<UsbSerialPort> ports = driver.getPorts();
}

I have a device with vendor id 0x2341 but driver always return null.

Thanks in advance.

kunofhan, 2018-10-29:
meet with the same problem

kunofhan, 2018-10-29:
Thank for your share,wait online

BrotherV, 2018-10-29:
Send an email an I'll reply it for you.  find my email in my profile.

kunofhan, 2018-10-29:
haven't see your email in your profile

> Send an email an I'll reply it for you. find my email in my profile.

haven't see your email in your profile.

kunofhan, 2018-10-29:
> haven't see your email in your profile
> 
> > Send an email an I'll reply it for you. find my email in my profile.
> 
> haven't see your email in your profile.

> Send an email an I'll reply it for you. find my email in my profile.

My email :956397053@qq.com , don't know whether you can send to me

BrotherV, 2018-10-29:
> haven't see your email in your profile
> 
> > Send an email an I'll reply it for you. find my email in my profile.
> 
> haven't see your email in your profile.

I've sent you my codes. if you had any problem, you know how to find me.

acaliaro, 2019-05-21:
@BrotherV could you share your work with me? Thanks

eclectice, 2019-05-21:
You can refer https://github.com/mik3y/usb-serial-for-android/issues/220 for usage idea and a useful example app from kai-morich fork.

acaliaro, 2019-05-21:
Thanks @eclectice I have found another solution

BrotherV, 2019-05-21:
@acaliaro Sure, give me your email and I can send it all for you.

acaliaro, 2019-05-21:
Thanks @BrotherV , I have found another good solution to this project. I will contact you if I will need your help in the future. Thanks again

BrotherV, 2019-05-21:
@acaliaro Your welcome. feel free and contact me with this email: mohsen_hatami@ymail.com
Good luck

kai-morich, 2019-10-26:
please look into examples at end of https://github.com/mik3y/usb-serial-for-android#quick-start for permission handling

closed 2019-10-26 by kai-morich

lythamhoa, 2021-04-26:
> I finally finished the project and I did it with two library(mik3y and felhr85 usb serial for android). If you would like, I can share what I did.

I need the for my project, Can you send me your code use mik3y or felhr85. Thanks. My email. thang.phamduc@gmail.com