mik3y/usb-serial-for-android #213: FTDI DT2232D issue

rueichang, 2018-12-11:
Hello everyone,
   
  I met an issue on the FT2332D.  It is a USB to dual COM chip.  So that I would have two USB nodes in my system(My kernel is 4.1.15).

  One of two nodes information from UsbSerialPort is /dev/bus/usb/001/053 device_id=1053 port_number=0
   And another information is from UsbSerialPort is /dev/bus/usb/001/055 device_id=1055 port_number=0

  Now it's okay to access 1055 COM, but 1053 cannot work.  I wonder if this issue happened due to the same port number.  

   Could you tell me how to change the port_number? Thanks in advance.

eclectice, 2018-12-11:
For multiport support see #212 kai-morich fork [kai-morich:multiport](https://github.com/mik3y/usb-serial-for-android/pull/212)

rueichang, 2018-12-17:
@eclectice  Thank you so much. We had tried your recommendation. And we're able to find another device, but there is still a little problem with it. Seems the source code of multiport is unstable. We are trying to fix it. Anyway, many thanks.

closed 2018-12-17 by rueichang