mik3y/usb-serial-for-android #214: Cp21xx freezes after 10 minutes of work in usbSerialExamples

tenkjm, 2018-12-12:
I try to work with example from this repository working on 115200 baud. But after aproximately 10 minutes the application freezes and the device list activity cannot see the device.

kai-morich, 2019-01-16:
I also implemented a service to keep the connection while the app is in background or rotating. To prevent the app beeing killed, you have to implementing a foreground service with a notification in the android status bar. This requires various interactions with the activity class, so this is best done on top of this library.
I am planning to post a stripped-down version of my 'serial usb terminal' on github containing such a foreground service

De5tr0yer, 2019-01-17:
If you don't want to use a service, you may circumvent the android auto-kill with automatic restart every once in a while.

My experience showed that the time-until-auto-kill varies and you can never be certain when your comm channel will be closed.

@eclectice gives you a complete answer, but I will still provide my solution with a ExecutorService
Please excuse my rudimentary explanation and lack of correct expressions, I am no expert in coding on such high level. I still managed to sicker through when creating a parser from my car steering wheel keys to a nexus 7.

II needed to use a service, which will also always be handled as foreground by android - it is just the nature of the system. Still, I did not want to have any activity active/visible, thus needed a sort of "background" service. You can achieve this by configuring the icon on the status bar to show only on pull-down. This will still "count as foreground" the whole time the notification is active.

In short steps what I've done to realize this:

Android Manifest: declare a service (Service.java); declare receiver (Receiver.java) on USB_DEVICE_ATTACHED. 

Receiver.java: onReceive creates a new Intent, which is then started through context.startService.

Service.java: 
onCreate gets USB broadcast, checks if the right device product ID is available and creates connection.

onStartCommand creates SerialInputOutputManager, which has a Listener defined and then sends it to ExecutorService as a single thread executor.

Listener has my desired function in "onNewData".

Hope it helps.

chaoderrick, 2019-02-01:
Hi Kai and De5tr0yer, thanks for your help. I was also planning on implementing a service down the line (not only due to the connection being killed after idle time, but also since the user can interact with the serial device when SerialConsoleActivity isn't the foreground activity, and when they return to that Activity their data isn't there).

Kai, I'll follow this thread and look forward to when you post your stripped-down version.

kai-morich, 2019-02-03:
please look here for my app including a foreground service: https://github.com/kai-morich/SimpleUsbTerminal

tenkjm, 2019-02-03:
Think you for solutions!   I managed to solve with forground services. But unfortunately its imposible to use Master usb and charge at the same time.  Now I have to use Windows 10 tab.

kai-morich, 2019-02-03:
i have a usb otg multi port hub with usb power input. charging + usb otg at the same time works only on samsung s7. with all other devices i can do only one of both.

closed 2019-10-06 by kai-morich