mik3y/usb-serial-for-android #215: How can i use this library in unity?

kamran1993, 2019-01-05:


eclectice, 2019-01-17:
You need to describe your use case for using a USB-Serial dongle in your Unity Engine.  I think there is no problem to use USB-Serial library in your Unity engine.  But, there is something you need to aware.

You will likely debug Unity engine on your test device through USB debugging mode (where you will use a USB cable to connect to your development PC to debug), therefore you won't be able to debug your USB-Serial dongle to check your codes that use this library until it is connected to your test device. 

Later, you will get some logcats of using USB-Serial dongle with any exceptions (if they happen) right after reconnecting in the USB debugging mode again. 

Or, you can use any crash reporting library to catch any exception (by saving some logcats) that can be sent over an email or stored into a folder to debug issues with USB-Serial library.

kai-morich, 2019-10-06:
compile this library to .jar with gradle task 'createFullJarRelease', then import into Unity

closed 2019-10-06 by kai-morich