mik3y/usb-serial-for-android #216: Does this works with Ionic 4?

weijyewang, 2019-01-14:
Does this works with [Ionic 4](https://beta.ionicframework.com/docs/)? Just wondering because I did not see any steps for Ionic in the documentation.

eclectice, 2019-01-17:
This library is just doing low-level USB-serial stuff. You need to create your own native Ionic wrapper (Cordova plugin) for this library if you can do that. Currently, Ionic only has native wrapper support for Bluetooth, Camera and a few other things where someone can connect to an IoT device; for example: https://blog.ionicframework.com/ionic-and-the-internet-of-things/

See for your self the native Cordova plugin codes for Camera support for Android:
https://github.com/apache/cordova-plugin-camera/blob/master/src/android/CameraLauncher.java

Here is the native Cordova plugin for Bluetooth Classic support for Android:
https://github.com/don/BluetoothSerial/blob/master/src/android/com/megster/cordova/BluetoothSerial.java

kai-morich, 2019-11-01:
moved to [wiki](https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#does-this-library-work-with-ioniccordovaxamarinapp-inventor-development-framework)

closed 2019-11-01 by kai-morich