mik3y/usb-serial-for-android #219: Which devices can charge while also connected as OTG Host?

chaoderrick, 2019-02-04:
@tenkjm, @kai-morich 

Sorry, not technically an issue with this library, but I invested weeks in building my app and only now am realizing the Nexus 10 cannot simultaneously charge while connected in OTG host mode. (Technically it can with a OTG Y cable AND a magnetic POGO charger - but the POGO charger is no longer available).

Kai mentioned the Galaxy S7. Has anyone else had luck with other Android devices (hopefully tablets)?

chaoderrick, 2019-02-05:
I found this list of Samsung devices: https://lavasimulcharge.com/nsts-ests-sts-rbm-compatibility-list/

Heavylama, 2019-05-17:
We also faced such a problem.

Our solution was to use a chinese tablet which has extra plug for external charging, this device allowed to be charged through this port and freed the microUSB port for simultaneous OTG usage. 

The device we used was a rugged 7,8" tablet running Android 7. I have no idea what model it was though, it only said "PAD", the designed seemed to be very similar to multiple other chinese devices. First revision had also external charging jack but didn't allow simultaneous charging, this was fixed in second revision.

derrickrc, 2019-05-17:
Thanks a lot! I ended up switching libraries to a serial/bluetooth one instead: https://github.com/harry1453/android-bluetooth-serial.

puwenjie, 2019-11-21:
you may be not need this devices
you can use virtualbox solve this problem.
I successfully used the computer to debug the android development OTG program.
[physical-otg-devices](http://scrumsofanarchy.com/interfacing-physical-otg-devices-to-genymotion-android-vm/#comment-96591)

olimaticer, 2023-03-11:
The issue can be solved by cheap USB-C adapters like this
USB-C --> OTG to
USB-C <-- Power
HDMI <-- Screen and sound
USB 3 <-- Data
You have to figure out the plug-in order of power and data.

https://www.tapatalk.com/groups/rfobasic/otg-multiport-adapter-t6432.html#p48332

closed 2023-04-01 by derrickrc