mik3y/usb-serial-for-android #220: working with FTDI 232

deey125, 2019-04-15:
Hi i need connect the cable Abacus Electrics Optical with chip FTDI 232 to Android 7, i am try with this library, but i don't get connection, if somebody could help me  this my E-Mail, deey125@gmail.com i need support customized

eclectice, 2019-04-16:
At least have the initiative to show your current code implementation. Try ProbeTable instance like below:

```
    //A runnable that attempts to open a serial connection to the USB device
    private class SerialPortOpener implements Runnable {
        @Override
        public void run() {
            final UsbManager usbManager = (UsbManager) getSystemService(Context.USB_SERVICE);

            //List<UsbSerialDriver> drivers = UsbSerialProber.getDefaultProber().findAllDrivers(usbManager);

            ProbeTable customTable = new ProbeTable();
            //<!-- 0x0557 / 0x2008: ATEN International Co., Ltd UC-232A Serial Port [pl2303] -->
            customTable.addProduct(0x0557, 0x2008, ProlificSerialDriver.class);
            //<!-- 0x0403 / 0x6001: FTDI FT232R UART -->
            customTable.addProduct(0x0403, 0x6001, FtdiSerialDriver.class);
            //<!-- 0x10C4 / 0xEA60: CP210x UART Bridge -->
            customTable.addProduct(0x10C4, 0xEA60, Cp21xxSerialDriver.class);

            UsbSerialProber prober = new UsbSerialProber(customTable);
            List<UsbSerialDriver> drivers = prober.findAllDrivers(usbManager);

            for (UsbSerialDriver driver: drivers) {
                if (driver != null) {
                    for (UsbSerialPort port: driver.getPorts()) {
                        if (port != null) {
                            mUsbDevice = port.getDriver().getDevice();
                            if (mUsbDevice != null) {
                                Log.d(UsbSerialService.TAG, Integer.toString(mUsbDevice.getDeviceId()));

                                if (usbManager != null) {
                                    mUsbConnection = usbManager.openDevice(mUsbDevice);
                                    if (mUsbConnection != null) {
                                        mSerialPort = port;
                                        if (!mUsbDevice.getDeviceName().equals("/dev/bus/usb/002/003")) {
                                            String deviceProp =
                                                    "  DEV = " + mUsbDevice.getDeviceName()
                                                            + ", VID = " + mUsbDevice.getVendorId()
                                                            + ", PID = " + mUsbDevice.getProductId()
                                                            + ", ID  = " + UsbSerialService.this.getProductName()
                                                            + ", MID = " + UsbSerialService.this.getManufacturerName()
                                                            + ", SID = " + UsbSerialService.this.getSerialNumber();

                                            Log.d(UsbSerialService.TAG, "valid usb serial device:" + deviceProp);

                                            if (MainApp.getActivity() != null && Constants.ENABLE_USB_DEVICE_INFO) {
                                                Tools.showMessageBar(MainApp.getActivity(), MessageType.INFO, deviceProp);
                                            }
                                        }
                                        break;
                                    }
                                }
                            }
                        }
                    }
                }
            }

            if (mUsbConnection != null) {
                if (mSerialPort != null) {

                    try {
                        mSerialPort.open(mUsbConnection);
                        mSerialPort.setParameters(mBaudRate, DEFAULT_DATA_BITS, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE);
                        mSerialPort.setDTR(true);
                        mSerialPort.setRTS(true);
                        Log.d(UsbSerialService.TAG, "successfully create connection to usb serial device: " + mUsbConnection.getSerial());
                        Log.d(UsbSerialService.TAG, "CTS - Clear To Send: " + mSerialPort.getCTS());
                        Log.d(UsbSerialService.TAG, "RTS - Request To Send: " + mSerialPort.getRTS());
                        Log.d(UsbSerialService.TAG, "DTR - Data Terminal Ready: " + mSerialPort.getDTR());
[... 54 lines cut ...]
            e.printStackTrace();
        } catch (IllegalAccessException e) {
            e.printStackTrace();
        } catch (InvocationTargetException e) {
            e.printStackTrace();
        }
        return result;
    }

    //android.hardware.usb.UsbDevice.getManufacturerName() is only available since API 21
    private @Nullable String getManufacturerName() {
        String result = null;
        try {
            Method method = android.hardware.usb.UsbDevice.class.getMethod("getManufacturerName");
            result = (String) method.invoke(mUsbDevice);
        } catch (NoSuchMethodException e) {
            e.printStackTrace();
        } catch (IllegalAccessException e) {
            e.printStackTrace();
        } catch (InvocationTargetException e) {
            e.printStackTrace();
        }
        return result;
    }
```

kai-morich, 2019-04-16:
please try with github.com/kai-morich/SimpleUsbTerminal. If no device is shown, your Android device might not support USB host mode. If a device is shown, but not recognized as FTDI, it might use a custom VID/PID that you have to add to your prober

eclectice, 2019-04-16:
@deey125 Ok, I have checked [Abacus Electrics ](http://www.abacuselectrics.com/index.htm) website and it seems like it sells optical cables. From [usb.id](http://www.linux-usb.org/usb.ids) database, it is based on FTDI chip (0x0403) with PID (0xF458). You just need to add the info in the `ProbeTable` instance:

```
//<!-- 0x0403 / 0xF458: ABACUS ELECTRICS Optical Probe -->
customTable.addProduct(0x0403, 0xF458, FtdiSerialDriver.class);
```

deey125, 2019-04-16:
> please try with https://github.com/kai-morich/SimpleUsbTerminal. If no device is shown, your Android device might not support USB host mode. If a device is shown, but not recognized as FTDI, it might use a custom VID/PID that you have to add to your prober

Hi, i am trying with this  application and the response is not driver :( but i tried with this code: customTable.addProduct(0x0403, 0xF458, FtdiSerialDriver.class); and  it's work but now i have other problem, the response now is: the endpoints are insufficient :(

deey125, 2019-04-16:
> @deey125 Ok, I have checked [Abacus Electrics ](http://www.abacuselectrics.com/index.htm) website and it sells optical cables with a few based on USB. From [usb.ids](http://www.linux-usb.org/usb.ids) database, it is based on FTDI chip (0x0403) with PID (0xF458). You just need to add the info in the `ProbeTable` instance:
> 
> ```
> //
> customTable.addProduct(0x0403, 0xF458, FtdiSerialDriver.class);
> ```

Thanks, your code work it but i now i have other problem, i am trying with:  https://github.com/kai-morich/SimpleUsbTerminal. but now i get the response: connection failed insufficient number of endpoints(2)

eclectice, 2019-04-17:
Can you try on another laptop or host device to check if the problem persists or not? Otherwise, it is possible a bug in determining the number of endpoints for your USB device that may require more for inputs and outputs to work properly which is insufficiently supported by the standard FtdiSerialDriver.class

deey125, 2019-04-17:
I was checking with 2 different devices however I discovered that the chip only has 2 endpoints and the library to open the port requires minimum 3

eclectice, 2019-04-17:
> I was checking with 2 different devices however I discovered that the chip only has 2 endpoints and the library to open the port requires minimum 3

Try to debug and make sure it invokes the correct driver FtdiSerialDriver instead of CdcAcmSerialDriver

eclectice, 2019-04-17:
I've checked the sample app; it has the class [CustomProber](https://github.com/kai-morich/SimpleUsbTerminal/blob/master/app/src/main/java/de/kai_morich/simple_usb_terminal/CustomProber.java) which only checks specific VID/PID to invokes CdcAcmSerialDriver. You may need to modify this code.

deey125, 2019-04-22:
Thanks the driver work it y can send frames to smart meter thank you very much

closed 2019-04-22 by deey125