mik3y/usb-serial-for-android #223: Communication freezes on final UsbRequest response = mConnection.requestWait();

emKaroly, 2019-06-13:
I am using the CDC driver the communication works both way fine between the device and a android app until the device should send data multiple times but i catch only the first chunk because the CDC freezes in read on line "final UsbRequest response = mConnection.requestWait();"

I am using the kai-morich branch 1.7.0 but no issues tracker there.

kai-morich, 2019-06-14:
please try with https://github.com/kai-morich/SimpleUsbTerminal

alexwan02, 2019-11-08:
SerialInputOutputManager#step would read first ,then execute write. We want to write first , and get read buffer, in this case , `step` is not a good design

kai-morich, 2019-11-08:
I do not recommend the usage of ``SerialInputOutputManager.writeAsync`` as it will block as you mentioned. Even if it would not block, the write would be delayed unpredictably somewhere between 0 and ``READ_WAIT_MILLIS`` msec.

I got best results with using ``SerialInputOutputManager`` for reading and directly writing in the UI thread.

Nevertheless I am looking into the code to restore the ``timeout`` handling in ``read``, but ``UsbRequest.requestWait`` only supports timeout starting with Android 8.0 and for older Androids the previous ``bulkTransfer`` approach would have to be used, but ``bulkTransfer`` definitely causes data loss at high baud rates.

kai-morich, 2019-11-15:
read timeout is supported again with e2e9df846386ccdbf6625c057fd1cd5f94c16d41

closed 2019-11-15 by kai-morich