mik3y/usb-serial-for-android #228: How to compile into jar?

codewise-nicolas, 2019-09-07:
Tried to compile from source but cant seem to figure out the "magic" command to get gradle to generate a jar for me.
Tried the usual build, compile, jar, etc to no avail.
Any assistance would be greatly appreciated - trying to add/fix setRTS/setDTS for the CP2102 and build a JAR file I can use with the Cordova plugin cordovarduino.
Thanks!

kai-morich, 2019-09-08:
For rts/cts handling you best look into Linux kernel

codewise-nicolas, 2019-09-08:
Yes thank you, I have an idea of what I need to add code wise.
I just need assistance figuring out what gradle commands to make it build a
jar library.

kai-morich, 2019-09-09:
my fork is available on jitpack. I thought this includes aar/jar file, but when checking yesterday I noticed it isn't included. For java/kotlin apps this is not an issue as gradle will compile the source dependency. I will check why libs are missing

codewise-nicolas, 2019-09-10:
Thanks, Im looking forward to adding rts/dts support for a project Im working on and then pushing the work back up so others can benefit too.
I just cant for the life of me figure out the gradle commands.
The reason I need a JAR is simplicity when integrating with cordova and the build process on that project, versus doing a dependency to the source and trying to build the two codebases at the same time.

kai-morich, 2019-09-13:
please try gradle task 'createFullJarRelease'

kai-morich, 2019-10-06:
compile to jar with gradle task 'createFullJarRelease' is documented here 
2bee5b930b2f7f50e50895ec6e6e5d70b4881cc2 -> usbSerialForAndroid/build.gradle

closed 2019-10-06 by kai-morich