mik3y/usb-serial-for-android #232: Reusing UsbRequest objects instead of creating new one

lukegluke, 2019-10-21:
Hi,

For now using library UsbRequestJNI is logging an init and close messages for every transaction over the USB interface (at least for CP21xx and FTDI)
https://stackoverflow.com/questions/28559394/suppress-usbrequestjni-alogd-log-messages

It can be filtered by "setprop log.tag.UsbRequestJNI SUPPRESS", but it seems it would be better to create mUsbRequest object in open() and reuse it in read() instead of creating new one on every read.

kai-morich, 2019-10-30:
with 9ea936b14a971edc5702f1f97fa61b0948351e01 I removed this usbRequest.cancel() logic and replaced it with releaseInterface(...) that was already was used in prolific Driver. With releaseInterface() the read() is interrupted also on older Android versions. I tested with CP2102 and CP2105. Please test with this version.

What is the use with 'Serial Usb Terminal'? There you can select a driver for unrecognized devices.

lukegluke, 2019-10-31:
Hello, thanks for feedback.
I've test this version but it doesn't help in all cases. Doing a few connection-reconnection still leads to stop responding to controlTransfer with CP2105 and CP2108, but it seems that unlikely usbRequest.cancel() logic now issue is related to some controlTransfers, not for all control requests.
Also this commit adds releaseInterface to ftdi driver too. I had ftdi 2232d and it starts hanging out on oppening (after closing) in SIO_RESET_REQUEST controlTransfers too. 

I'm testing on Android 8.1.
What can happened if i omit releaseInterface? Possible run out of descriptors? Or just not release exclusive access and other apps could use it (this is acceptable to me in my project)?

By 'Serial Usb Terminal' I meant [Android Serial USB Terminal app](https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal)

kai-morich, 2019-11-01:
I know *Serial USB Terminal*, it's my app. What do you mean with: *Serial Usb Terminal doesn't allow to work with not configured device*?

The prolific driver already had releaseInterface(), the other drivers didn't, so I expect it works both ways. As releaseInterface() more reliably terminates the read() method called by SerialInputOutputManager thread, I added it to all drivers. As next step I will use this version in *Serial USB Terminal* and have a look at the Google Play error / crash reports.

I never opened multiple ports in parallel and I do not remember error reports about devices hanging after multiple connects.

lukegluke, 2019-11-02:
I mean that Serial Usb Terminal write "Connection failed: Error setting baud rate" on serial port opening if controlTransfer failed. In my application I ignore controlTransfer failures and assume success opening and further work with read and write (bulkTransfers) works fine. I just mention it to specify that the problems are only with controlTransfers. Sorry for confusion.

About multiple ports. In fact I got these hang outs and connection failed error even when opening **one** port of CP2105 or CP2108 in Serial Usb Terminal just after a few switches between different ports of same usb device.

lukegluke, 2019-11-24:
Returning to topic.
UsbRequestJNI init / close messages make a real mess in logcat, can't use it anymore. It's a real issue.
Reduce the tracing level through setprop log.tag.UsbRequestJNI info is not an option, because it resets after the reboot.

kai-morich, 2019-11-27:
To my understanding the logcat messages are annoying while developing, but not an issue when using the app?

lukegluke, 2019-11-27:
While developing and supporting (troubleshooting)

kai-morich, 2019-12-13:
implemented with ce97a3408bf37e5e7052be2283ebe2e747d6577f

lukegluke, 2020-02-03:
Hi, sorry for the late reply.
Thanks for your work. Tested ce97a34 as is for some time.
I use this driver through jni in c++ project and open all available ports simultaneously (SerialIoManagers are executed using newCachedThreadPool). 
By the way I didn't find any evidence if UsbDeviceConnection is thread safe, is it really right to open same usb device several times with openDevice?

Driver works fine with four serials at once on CP2108 and single FT232R. CP2105 also works, but after several fast openings-closings it totally hangup calling mConnection.bulkTransfer on the first writing to second port. Changing timeout to 2s instead of 0 stops total hangups, but it looks like there are freezes on these 2s (but if fact it sends something right, do not test for writing a lot data).

Previous version works good for all usb, but I use it without releaseInterface on close.
Try to return releaseInterface to old version - yeah it hangups similarly to new version.
But removing releaseInterface from new version do nothing - hangups on bulkTransfer. 
Maybe it is all about hardware in fact, do not know, just described the issues I have.

kai-morich, 2020-02-03:
maybe you have to put some lock into `openInt` to avoid alternating USB initialization requests.

lukegluke, 2020-02-07:
I was not really correct. In fact ports are open one by one, there is no interference on opening, I've confirmed that, so lock is not the solution as I see. After opening, ports then are read simultaneously, yes. Also forgot to mention that if I disconnect usb it exits from hangup in bulkTransfer and can work correctly if usb connected once again.

kai-morich, 2020-02-07:
can you try with timeout=0 which causes CommonUsbSerialPort::read() to use queue + requestWait?

lukegluke, 2020-02-07:
I use zero timeouts everywhere already (I had just tested non zero timeouts when found a problem). 
Hangup is in mConnection.bulkTransfer on **writing**, not reading.

lukegluke, 2020-03-09:
Hi @kai-morich,

I've made one more attempt to use new driver version and several test with it and previous too.
And now I've found out that those "_hangups on calling mConnection.bulkTransfer on the first writing to second port_" were in fact my bad.

The main issue that provoked me to add more confusion is fails with controlTransfer (timeout) I mentioned previously. I thought that it is hardware issue, so I corrected original driver and didn't throw exception from setConfigSingle and continued to work with port anyway. Also I removed releaseInterface on closing, because it leads to fails with controlTransfer on next opening.
And this "frankenstein" worked quite stable somehow.

But my bad was that besides these corrections, to reduce controlTransfer timeouts I also added additional check into setConfigSingle so that if controlTransfer failed (timeout) at least once it will not called again. I made this check as the driver (not port) variable.

Now I've found out that controlTransfer in fact starts failed (after several open/close) not as the function itself, but only on first port CP2105 (zero indexed) and only for IFC_ENABLE (0x00) and SET_MHS (0x07) request codes:
```
//opening
E Cp21xxSerialDriver: Control transfer for port 0 failed: 0 / 1 -> -1
E Cp21xxSerialDriver: Control transfer for port 0 failed: 7 / 771 -> -1
...
//closing
E Cp21xxSerialDriver: Control transfer for port 0 failed: 0 / 0 -> -1
```
Despite these errors, first port continue to work properly without problems (like it properly was turned on and set before). Also SET_BAUDRATE (0x1E) and SET_LINE_CTL (0x03) control transfers get written fine and really change RS232 params.

All control transfers to second port also work fine. But my foolish idea to not call controlTransfer after first fail in driver (in fact, on first port), left the second port in not turned on and unconfigured state, that's why (as I think) my app totally hangups in mConnection.bulkTransfer on the first writing to second port until USB was deattached.

I've tested previous version of driver and it fails and hangups the same way in the same circumstances. Now I use new driver without my foolish check even with releaseInterface - it works. controlTransfer fails is still the issue, but as workaround I ignore fails of serialConfig and reduce controlTransfer timeout time to 500ms.
Anyway it is not related to reusing UsbRequest and this particular issue. More over your original code in this case would not allow to open the first port due to setConfigSingle exception.

Sorry for huge comment and for the previous mess and confusions.
I close this issue, thanks for the work!

p.s. on 4port CP2108 controlTransfer timeouted sometimes not only on first port, but in addition also on the second and even on third ports but rare.
And no error messages on single port CP2102.

closed 2020-03-09 by lukegluke

lukegluke, 2020-03-17:
But still I hurried with conclusions. In my app I have two parts - one where all serials are opened and one with update process, where serials opened one by one. I found that second process not work - does not send nothing. After this, if first process with opening all serials is run, it leads to total hangup like before, until usb is detached. One aspect - I use driver without releaseInterface. Did not try with it, unfortunately for now I don't have more time to investigate this. I revert back to previous version of driver (without reusing UsbRequests) - it works for me in all my cases (I use driver without releaseInterface also).

This is some info, just in case someone will be playing with using several ports together and encountered with same problem (or not, and will be kind enough to explain to me how to do this properly).