mik3y/usb-serial-for-android #235: ERROR: Failed to resolve: com.github.mik3y:usb-serial-for-android-Tag

mahaju, 2019-11-07:
I have added maven { url 'https://jitpack.io' } inside repositories in the root level build .gradle file

In the app level build.gradle I added implementation 'com.github.mik3y:usb-serial-for-android:Tag' inside the dependencies{} block, at the end of everything that was already in there

After syncing gradle I am getting the error "ERROR: Failed to resolve: com.github.mik3y:usb-serial-for-android-Tag

What can I do to fix this? Currently the java code is only the default "hello world" inserted when creating a new android studio project. I have not changed anything else, just trying to add the USB library first.

kai-morich, 2019-11-07:
you have to specify a concrete tag, e.g. **master** or **2.0.0** as you can see here: https://github.com/kai-morich/SimpleUsbTerminal/blob/master/app/build.gradle

closed 2019-11-07 by kai-morich

kallenosf, 2022-05-12:
Thanks for this library.
I'm getting the same error.
I have included " implementation 'com.github.mik3y:usb-serial-for-android:3.4.3' " in dependencies{} in build.gradle(:app)

bhanu9009, 2022-09-06:
I was facing the same error
changed to the below configuration and the issue solved 
upgrading
android gradle plug-in version to 7.2.1
gradle jdk version to 11.*

and modifying below lines app/build.gradle solved the problem

compileOptions {
    sourceCompatibility JavaVersion.VERSION_11
    targetCompatibility JavaVersion.VERSION_11
}

kai-morich, 2022-09-06:
By changing versions you triggered a full rebuild. By using clean/rebuild/... commands in Android Studio should have accomplished the same effect