mik3y/usb-serial-for-android #238: there is function to readLine of incoming data?

tomeroto, 2019-11-13:
Thank you for your code.
Btw I have issue, I know you have set listener to receive data by callback "onNewData".
But do you have instant method to receive data that ending by some characters like new line "\r\n"?
So it will reduce wrong data to retrieve.

kai-morich, 2019-11-13:
this currently not part of the library, but could be implemented as Adapter on top of ``SerialInputOutputManager.Listener`` to buffer incoming data and return only complete lines with ``onNewData``

fab1o978, 2019-12-19:
> this currently not part of the library, but could be implemented as Adapter on top of `SerialInputOutputManager.Listener` to buffer incoming data and return only complete lines with `onNewData`

Any tips about how to do this? Thank you very much!

tobychui, 2021-07-20:
This thread is a bit old, but I ran into the same problem recently. That is why I implement the function mentioned above. Here is my gift to those who is looking for a solution to copy and paste in the future.

First you will need to implement your own custom listener like this:
```
private class CustomListener implements SerialInputOutputManager.Listener {

        @Override
        public void onNewData(byte[] data) {
            ReadLineHandler(data);
        }

        @Override
        public void onRunError(Exception e) {
           //Error will return when the port closed. Ignore this
        }
    }
```

Next, create a function to handle the incoming serial byte[] data. Here is an example for such line spiting function.

```
private String readBuffer = "";
// . . .
private void ReadLineHandler(byte[] data) {
      String incoming = new String(data, StandardCharsets.US_ASCII);
      readBuffer += incoming;
      while(readBuffer.length() > 0 && readBuffer.contains("\n") == true){
          String thisLine = readBuffer.substring(0, readBuffer.indexOf("\n")).trim();
          //Handle this line here

         //Trim the processed line from the readBuffer
         readBuffer = readBuffer.substring(readBuffer.indexOf("\n") + 1);
      }
}
```
Note that the while loop in the above function handle the case where more than 1 new lines are passed in from one listener event.

And it can be used like this:
```
CustomListener listener = new CustomListener();
usbIoManager = new SerialInputOutputManager(port, listener);
usbIoManager.start();
```