mik3y/usb-serial-for-android #239: Maximum number of bytes that can be sent or received at one time

mahaju, 2019-11-14:
When sending a very large byte[] array using the 
port.write(...) function, I don't receive all the bytes at the receiver
Even thought the bytes that are received are all correct, the number of bytes received are always different
For example, if I attempt to send a 60 byte array, sometimes the receiving android phone receives 19 bytes and sometimes 50 bytes in the onNewData(byte[] data) function

Is there a way to determine what the maximum possible number of bytes that can be sent reliably at one time? Does this depend on the exact USB-serial chip used?

I am using FTDI (I think FTDI 232) based USB to serial cable to connect two smartphones together. A long time ago (I don't remember all the details) using the same USB-serial cable I was able to communicate between a PC and a DSP chip sending out serial data, and I was able to send more than 80 bytes at one time reliably. Will this be different when trying to do the same thing on android? Is this a limitation of the driver or the FTDI hardware? Would switching to FTDI supplied drivers let me send and receive larger number of bytes at one time?

kai-morich, 2019-11-14:
It's not guaranteed that all data arrives in one packet, see https://github.com/mik3y/usb-serial-for-android/wiki/Troubleshooting#when-reading-a-message-why-dont-all-bytes-arrive-at-the-same-time

mahaju, 2019-11-14:
In my case when I receive less number of bytes than what I sent, the remaining data just seems to get lost. if I understood the post you linked to correctly, when I send 60 bytes and received 50, I should soon receive 10 more (which I means onNewData(byte[] data) would be called again with 10 bytes as received data?). Is this correct? Is there a maximum number of bytes that is always guaranteed to be received at once and is there a way to find out what it is? Or is it different every time?

kai-morich, 2019-11-14:
there is no guarantee how much bytes you get in one call. it also highly depends on the used chip. All chips have slightly different logic, to send data after some idle time or a buffer fill threshold.

You should not use a too small read-timeout, as this can cause data loss somewhere in the Android USB stack. a timeout of some hundred milliseconds is ok

mahaju, 2019-11-14:
That's a parameter for the UsbSerialPort class read(byte[] , int) function isn't it? I am not using that function but I am only using the onNewData(byte[] data) function (shown in the example code in the github page) which is called automatically when data is received. Does this mean once I receive part of the data automatically I need to manually call this read() function until I get all the remaining data?

kai-morich, 2019-11-14:
onNewData() will be called again with more chunks. you don't have to call read() manually

closed 2019-12-08 by kai-morich