mik3y/usb-serial-for-android #241: Split example applications and main library

tva-TIS, 2019-11-14:
The examples beeing located in the same directory as the library itself makes it harder to contribute to the library, as the repository cannot be checked out into another android project. Therefore the changes need to be tested within the example application, or the changes need to be manually included.

Solution: Create a different repo for the examples that include this repo (which then should be the root folder of the library) as a submodule.

kai-morich, 2019-11-14:
this [wiki](https://github.com/mik3y/usb-serial-for-android/wiki/Build-Variants#local-maven-repository) mentions a setup for local developing and testing of usb-serial-for-android. would that work for you?

tva-TIS, 2019-11-15:
Oh sorry, i looked into the wiki but oversaw that.

closed 2019-11-15 by tva-TIS