mik3y/usb-serial-for-android #242: Does this library support RTS/CTS handshake?

mahaju, 2019-11-18:
I need to interface with a hardware with a USB-serial interface based on FT232RQ chip

It says in the hardware datasheet datasheet that it uses hardware CTS/RTS flowcontrol. However, when I call the port.setRTS(true) function, and immediately view the value of port.getRTS(), it stays at false.

Am I doing something wrong? I found this discussion (https://github.com/mik3y/usb-serial-for-android/issues/13) but it is from 2013 and it'a not clear whether RTS/CTS is implemented or not. 
Does this library support RTS/CTS flowcontrol?

kai-morich, 2019-11-18:
CTS/RTS is not supported for all drivers. For FTDI is missing, but there is an old PR #50
If you can rebase and test, a new PR would be welcome

kai-morich, 2019-11-23:
solved with #118

closed 2019-11-23 by kai-morich