mik3y/usb-serial-for-android #244: Prefer a more liberal open source license

satur9nine, 2019-11-20:
Too bad this project is LGPL. Restrictions such as this aren't really feasible for most developers these days:

> b) Use a suitable shared library mechanism for linking with the
>     Library.  A suitable mechanism is one that (1) uses at run time a
>     copy of the library already present on the user's computer system,
>     rather than copying library functions into the executable, and (2)
>     will operate properly with a modified version of the library, if
>     the user installs one, as long as the modified version is
>     interface-compatible with the version that the work was made with.

kai-morich, 2019-11-20:
I would also prefer a more permissive license. Probably @mik3y has chosen LGPL as his FtdiDriver is  to some extend a rewrite of libftdi and therefore he has choosen the same license. 

[Relicensing](https://en.wikipedia.org/wiki/Software_relicensing) might be some effort if done correctly, e.g. getting consent from major contributors and rewriting libftdi based code, using other sources for the FTDI specific logic, like baud rate calculation.

Since version 2.0.0 this project is available as library at jitpack, so you can use it as library instead of copying the sources into your project as was necessary before. The other LGPL requirement to make a library replaceable is difficult to achieve an Android as an app is packed with all libraries into an .apk file. Repackaging the .apk is recommended in other discussions, but I do not know how good this works.

mik3y, 2020-01-12:
> Probably @mik3y has chosen LGPL as his FtdiDriver is to some extend a rewrite of libftdi and therefore he has choosen the same license.

That's correct; the project was originally built for the FTDI driver, which I adapted from the LGPL libftdi & kernel driver sources. 

I'd be fine with you relicensing the core of the project (eg the non-driver framework) around something more permissive eg MIT. In turn, for places where GPL licenses are problematic, that would allow you to distribute the driver(s) separately & clients could choose whether or not to include them.

kai-morich, 2020-01-12:
Do you think it would be sufficient to rewrite the FTDI driver and relicense the full library under MIT license? 
A rewrite might loose support for some very old FTDI devices, but that would be ok for me.
I haven't investigated if the control commands required for FTDI and other devices are publicly available with a clear license. To me looks like most commands were reverse enginered a long time ago and since then used in various projects with many different licenses.

mik3y, 2020-07-02:
> @mik3y: Do you think it would be sufficient to rewrite the FTDI driver and relicense the full library under MIT license?

Hey @kai-morich - I'm sorry to answer this so late! That sounds right to me - I'm not a lawyer of course,  but it's only the FTDI bits that were originally GPL.

(And great job keeping the library maintained, dude! You've improved it a ton.)

kai-morich, 2020-07-04:
FTDI does not provide full documenation, but
* important parts like baud-rate formula and controltransfer bits are scattered over their documents
* wireshark allows revers engineering
* I found at least 5 other FTDI implementations 

so looks possible to rewrite the FTDI driver

kai-morich, 2020-07-05:
@mik3y various files have a comment `Copyright 2011-2013 Google Inc.` you added that comment already with the initial commit. What is copyrighted by Google and would it have an impact on changing the license?

kai-morich, 2020-07-27:
in v3 branch you can find a version with rewritten ftdi driver and MIT instead of LGPL library

@mik3y: please have a look at the copyright information in the MIT license, filled with best guess, as slightly different copyright notices and dates were found in sources and README.md

felixhaedicke, 2020-07-28:
I agree to relicense the code I contributed to the MIT license.

mik3y, 2020-07-29:
@kai-morich the original copyright is because I worked at Google at the time & they asserted copyright as a contingency of opensourcing. I left Google after fe2eb82 - and I am happy for you to relicense any of my work under MIT.

andreasb242, 2020-07-30:
I agree to relicense the code I contributed to the MIT license.

kai-morich, 2020-08-01:
v3.0.0 is available with rewritten FTDI driver and MIT license

closed 2020-08-01 by kai-morich

satur9nine, 2023-02-16:
Thanks everyone involved! I had actually forgotten about making this request but a new project came up and I see this is now MIT License!