mik3y/usb-serial-for-android #246: I can't get data instantly if the data is coming less than 1s interval,by FTDI Serial Driver

huijimuhe, 2019-11-21:
hello,I do the test of sending data(0xFF 0xFF) to my APP by 500ms interval(or any time less than 1 second), but I can't get the data instantly. There is a delay, it seems I can only get them by about 1s interval, when I stop the data sending, the data is still coming till all been received.I only tested FTDI Serial Driver wire, it works well on computer. 

Why is this  happen? I can't find answer in Android API document or here. Does any one know why? and how can i fix this?

thanks.

kai-morich, 2019-11-21:
for FTDI devices one can configure a buffer threshold before an USB request ist send, but this is not used by this library so you should get the data immediately. Does it work with some existing app like [SimpleUsbTerminal](https://github.com/kai-morich/SimpleUsbTerminal)?

huijimuhe, 2019-11-24:
no, simpleusbtermial is same performance.Guess there is no fix to this.Limited by the hardware or interface.

closed 2019-11-24 by huijimuhe