mik3y/usb-serial-for-android #251: Error setting up device: Init failed: #2

mayurdo, 2019-12-07:
Hi, 
I am getting below error Please help me on this
Device details : 
vid: '1a86',
pid: '7523',
driver: 'Ch34xSerialDriver'

Error :

2019-12-08 02:17:44.022 14762-14762/com.hoho.android.usbserial.examples E/SerialConsoleActivity: Error setting up device: Init failed: #2
    java.io.IOException: Init failed: #2
        at com.hoho.android.usbserial.driver.Ch34xSerialDriver$Ch340SerialPort.initialize(Ch34xSerialDriver.java:187)
        at com.hoho.android.usbserial.driver.Ch34xSerialDriver$Ch340SerialPort.open(Ch34xSerialDriver.java:120)
        at com.hoho.android.usbserial.examples.SerialConsoleActivity.onResume(SerialConsoleActivity.java:162)
        at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1413)
        at android.app.Activity.performResume(Activity.java:7408)
        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3855)
        at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3895)
        at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:51)
        at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:145)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:70)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1839)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:201)
        at android.app.ActivityThread.main(ActivityThread.java:6861)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:547)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:873)

kai-morich, 2019-12-08:
do you use a very old ch340?

mayurdo, 2019-12-09:
You mean old ch340 device or old Ch34xSerialDriver.java code file ?

I have downloaded project from "Clone or download" option 
I have weighing machine with serial port output, I have connected serial port to usb converter and then usb to OTG converter to connect with my mobile.

I have use the weighing machine with .NET C# code  and it's working ok with same serial port to usb converter to connect with my PC.

kai-morich, 2019-12-09:
I meant very old ch340 device. 
To my knowledge all ch340 drivers are based on reverse engineering. This library uses the same initialization sequence as the linux kernel driver. Other libraries use slightly different initialization, but as this might cause issues with other ch34x devices, I better keep as is.

mayurdo, 2019-12-09:
I have used below cable converter 
https://www.amazon.in/gp/product/B019JZAYCA/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I am not able to check this is old one or not.
Can you able to identify this converter is old ch340 device.

kai-morich, 2019-12-09:
hard to tell, but a device shipping with CD and claiming Windows 98 compatibility might still use old CH34x chip

mayurdo, 2019-12-11:
OK, 
I have ordered new one. Will try with that and update you accordingly.
Thank your for your help.

https://www.amazon.in/gp/product/B0756W8GXV/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

mayurdo, 2019-12-22:
Hi, 
The last purchase Connect also not work. 

But I have purchase another one with Prolific driver and that is working now.
Thanks for your help

boyjeon, 2020-02-03:
I modified the file 'Ch34xSerialDriver.java' as shown below.

private int controlOut(int request, int value, int index) {
			final int REQTYPE_HOST_TO_DEVICE = 0x40;  //0x41->0x40
			return mConnection.controlTransfer(REQTYPE_HOST_TO_DEVICE, request,
					value, index, null, 0, USB_TIMEOUT_MILLIS);
}

And the cable had normal communication with the device.

kai-morich, 2020-02-03:
the current value of 0x41 is really strange. When combining the different UsbConstants the correct value is 0x40. Looks like some devices are stricter. 

I changed the constant here: 2a77ebf8b9d3c651632c3d2d18a2386c0fddb1d3

kai-morich, 2020-02-12:
included in v2.2.1

closed 2020-02-12 by kai-morich