mik3y/usb-serial-for-android #252: ERROR: Failed to resolve: com.github.mik3y:usb-serial-for-android:Tag

garaninsoft, 2019-12-12:
Hi!
Android Studio 3.5.3
Build #AI-191.8026.42.35.6010548, built on November 15, 2019
JRE: 1.8.0_202-release-1483-b49-5587405 x86_64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
macOS 10.15.1

I try Quick Start:
1. Add jitpack.io repository to your root build.gradle.
2. Add library to dependencies.

But gradle allways throw exception: 

ERROR: Failed to resolve: com.github.mik3y:usb-serial-for-android:Tag

Help me please!?

kai-morich, 2019-12-12:
you have to specify a specific Tag, e.g. _master_ or _2.1.0_

garaninsoft, 2019-12-12:
... but don't work:

dependencies {
    implementation fileTree(dir: 'libs', include: ['*.jar'])
    implementation 'androidx.appcompat:appcompat:1.1.0'
    implementation 'androidx.constraintlayout:constraintlayout:1.1.3'
    testImplementation 'junit:junit:4.12'
    androidTestImplementation 'androidx.test.ext:junit:1.1.1'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.2.0'

    //implementation 'com.github.mik3y:usb-serial-for-android:2.1.0'
    implementation 'com.github.mik3y:usb-serial-for-android:master'

}

what i do wrong?

Best regards!

kai-morich, 2019-12-12:
maybe you have to sync the project

garaninsoft, 2019-12-12:
[image]
but, i do...

garaninsoft, 2019-12-12:
I forgot to return to the place after various experiments:

allprojects {                                                                              
    repositories {                                                                        
        google()
        jcenter()
        maven { url 'https://jitpack.io' } <------- this!!!
    }
}

Your answer: "you have to specify a specific Tag, e.g. master or 2.1.0" - is right
Thank's!!!

closed 2019-12-12 by garaninsoft