mik3y/usb-serial-for-android #254: GetStarted Example...

garaninsoft, 2019-12-20:
...This question is some continuation of my previous question.

In [git clone project](https://github.com/mik3y/usb-serial-for-android.git) I edit file 
`device_filter.xml`

by adding next vid/pid:

```
<?xml version="1.0" encoding="utf-8"?>
<resources>

...

    <!-- 0x2F5A / 0x0001: POS Asino V37 -->
    <usb-device vendor-id="12122" product-id="1" />
</resources>
```

But application don't see my device. Can you help me, what's wrong?

Best Regards!

kai-morich, 2019-12-21:
the `device_filter.xml` is only responsible to get your App auto-started when a device is connected. 

To get the right driver you additionally have to use a custom prober or enhance the `getSupportedDevices` method in the related driver class.

garaninsoft, 2019-12-24:
Thank you. That's exactly what I did: custom probe. Everything works.

closed 2019-12-24 by garaninsoft